JejuSoul/OBD-PIDs-for-HKMC-EVs #6: The VMCU codes for the Ioniq EV need to be confirmed.

JejuSoul, 2018-04-17:
The VMCU codes are bring created by ITStromer on the German Forum.

I am reading the German forum - via Google translate.
https://www.goingelectric.de/forum/hyundai-ioniq-allgemeines/obd-diagnose-mit-torque-t21111-100.html
and see this -
"That this is obviously not the temperature values I noticed yesterday and today. The values sometimes also jump funny. Currently I log the values of the PIDs longer, maybe there are any patterns. Have therefore taken out of the last post, the PIDs, so that no one is unsettled."

Can anyone confirm that the code for motor temp is valid . Otherwise I think I will have to remove it entirely

JejuSoul, 2018-04-18:
I have deleted the code for motor temp.
I think the problem with duplicate codes may have been a copy and paste error on my part.
But we cannot know because ITStromer has removed his original post.
Either way he removed the post because the codes may be wrong, so I will remove them too.

JejuSoul, 2018-04-23:
Yes. I too have been looking at the VMCU temperatures by looking at kinnison's data.
I believe the original 2 values by ITStromer were almost correct.
The letters used by Torque were slightly out. (Which might have been my error not his.)

Can you check these 2 values to see if they are temps.

003_VMCU Motor Temperature,Motor temp,2102,o-40,-40,125,C,7E2
003_VMCU MCU Temperature,MCU temp,2102,t-40,-40,80,C,7E2

kinnison, 2018-04-23:
I will try those later tonight if I can.  The car will be sat all day, so in theory the values should match the outside air temperature :-D

JejuSoul, 2018-04-23:
Here's one more to try

003_VMCU Real Vehicle Speed,Real Speed,2101,((o<8)+n)/10,0,180,km/h,7E2

The 2 bytes after the accelerator pedal depth PID are always zero when the car is stopped.
On the Soul EV real speed PID is a single byte immediately after the accelerator pedal depth.
But on the Ioniq we seem to have a 2 byte number divided by 10.
It is also Little Endian, ie the lower part comes first.

kinnison, 2018-04-23:
Oooh, I'll try and get results for that too.  You think it's in km/h though, not mi/h ?

kinnison, 2018-04-23:
Okay, I added the VMCU motor and MCU temperatures, and the vehicle speed decodes to my app and replayed the basic trace I provided on a different issue.
With that, I'm seeing the motor and mcu temperature rise in concert toward the start of the drive from an initial 6c to about 16c and then hold there.  the mcu temperature fluctuating by a degree up or down from time to time, but the motor temperature remaining constant.  Across the whole of the basic drive route, (about 6 minutes, top speed around 30mi/h) the motor temperature remained consistently about 16 degrees if that is to be believed until the last 30s or so of the drive where it fluctuated up and down by a degree instead of the MCU temperature.  I'm not entirely convinced *yet* of those.

The vehicle speed looks to me like it's (a) signed and (b) more likely (for me) mi/h and /100 not /10.
I see values up to around 3200 at the fastest part of my journey which would correspond to 32mi/h on the car's indicator which is about 30mi/h in the real world.  With that, it matches how I was previously estimating the vehicle speed (motor RPM seems to correspond approximately with speed in mi/h times 100 too, though motor RPM is in 100 jumps, not like the speed you found in the VMCU)

In conclusion, I'd say that the temperatures are still suspect, but the speed looks good if it were mi/h and /100.

JejuSoul, 2018-04-24:
Here's an image from a 30kWh Soul EV showing the VMCU Motor Temperature and Real Vehicle Speed

https://s5.postimg.cc/sedo6po43/acceleration_test.png

Could you post something similar using your values for temps, speeds and rpm from an Ioniq EV.
It doesn't seem right to me that a Korean ECU would use mph internally not km/h

JejuSoul, 2018-04-24:
thanks for the data.
I put the speeds into an excel chart just to check that we are really seeing driving speed.
It is divided by 100 as you suggest.
Here's the chart. - https://abload.de/img/speedgraphfau5b.png

Can someone in a country that uses km/h for speeds confirm that this equation gives a result in miles / hour.
003_VMCU Real Vehicle Speed,Real Speed,2101,((o<8)+n)/100,0,180,km/h,7E2

migle, 2018-04-25:
I haven't replied much, but I've been reading all that you post. You are moving so fast, that I can't post that fast. I'm from southern europe (km/h, no cold climate pack). I made a big trip on Monday and logged MCU temperature (unfortunately, not the other VMCU would-be temperatures).

[tojejusoul-sevilla.log](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/1945914/tojejusoul-sevilla.log)

I also logged the other temperatures and the Real Speed yesterday according to formula ((o<8)+n)/10. I don't think it's mph. If it's a speed, I would just plot that with gnuplot and it would be visible. But I don't have the time right now, so I leave the discovery to you.... Speed is really useful for performance measurements (drag coefficient, etc).

[tojejusoul-speed.log](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/1945919/tojejusoul-speed.log)

And the (cut) full log:

[tojejusoul.log](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/1945920/tojejusoul.log)

JejuSoul, 2018-04-25:
Mojo in Sweden just confirmed that speed is in mph -  miles per hour. - http://www.ioniqforum.com/forum/234-hyundai-ioniq-electric-ev/9962-setting-up-torque-show-bms-data-ioniq-ev-2.html.

Migle - I updated the formula so that it is divided by 100 not 10.
003_VMCU Real Vehicle Speed,Real Speed,2101,((o<8)+n)/100,0,180,mph,7E2
I will look at your log data later, thanks for posting it.

migle, 2018-04-25:
I could read the log again. Yes, it looks like mph. I had missed the fact that I was recording GPS speed in meters/sec, not km/h.

JejuSoul, 2018-04-25:
Migle  - I just looked at your log data.
It confirms:
1/ The formula is mph
2/ The value for real speed exactly matches GPS speed.
3/ If you can confirm that at one point you reversed, then the formula is signed.

I plotted your data here - but removed a few 654 km/h values which are actually 1 km/h in reverse.

https://abload.de/img/speedgraph276r50.png

IoniqBEV, 2018-04-30:
Hi Jeju,
I use Torque pro in my Ioniq BEV, all works OK- only PIDs in 003_VCMU not. All values is negative and without changes...

migle, 2018-04-30:
Check your OBD2 dongle. Is it good? Is your reading of SoC Display good? How does it compare to SoC BMS?

JejuSoul, 2018-04-30:
IoniqBEV thanks for posting. Can you post details of your adapter and the response it gets from  ELM327 Identifier - https://play.google.com/store/apps/details?id=com.applagapp.elm327identifier

There have been issues before that some adapters are not capable of reading all PIDs
see -  http://www.mykiasoulev.com/forum/viewtopic.php?f=6&t=711&p=7093#p7091

IoniqBEV, 2018-04-30:
Thanx You very much, Your info was usefull. I had blue adapter, today I buy white- with him working all PIDs. I add pictures for You info.
[image]
[image]
[image]

IoniqBEV, 2018-05-04:
Hi Jeju,
Torque do not know read distance from Ioniq, I use formula to calculate trip distance, You can add to extragauges:
004_Trip_Distance,DST,,(val{Average trip speed(whilst moving only)}*val{Trip time(whilst moving)})/3600,0.0,1000,km,7E4
Average speed and time Torque calculated from GPS data, if GPS signal lost (in tunnel), calculated distance have difference. Now I tested distance calculated from Real vehicle speed...

IoniqBEV, 2018-05-06:
These two extra values are standard PIDs in TorquePro, only must be entered in origin name (english- as I wrote). If user switch Torque in his language, PIDs may be translated. ;-)

JejuSoul, 2018-05-07:
I have never used these built in codes. And there is so little documentation.
How do you set these codes to use  Real vehicle speeds instead of calculating speed from GPS data.

kinnison, 2018-05-10:
I wonder if the GPS coordinates that the car knows from its GPS would be available somewhere.

migle, 2018-05-10:
So, about temperatures, MCU and motor temperature, those don't appear to be right. Motor temperature is weird and MCU temperature doesn't look like a temperature. On a long trip, MCU temperatures are between 7 and 16. The offset could be wrong, but it is sometimes 10 when it should be hot, it also drops suddently to 7....

Here's some data.

[tempstojejusoul.txt](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/1992535/tempstojejusoul.txt)

IoniqBEV, 2018-05-10:
Hi Jeju, sorry for delay.
Here are my PIDs calculate with Real vehicle speed:
004_CALC_Average_Speed,AVR SPD,,1.6*val{004_Sum_Speed}/val{004_Count_Speed},0.0,150.0,km/h,7E4

Average speed is whilst moving only, constant 1.6 is convert mph to km/h
Formula need four temporary variables:

004_Sum_Speed_Tmp,SumSpdTmp,,val{004_Sum_Speed}+ABS(val{003_VMCU Real Vehicle Speed}),0,3600000,mph,7E4
004_Sum_Speed,SumSpd,,val{004_Sum_Speed_Tmp},0,3600000,mph,7E4

004_Count_Speed_TMP,CountTmp,,val{004_Count_Speed}+LOOKUP(ABS(val{003_VMCU Real Vehicle Speed})::0=0:0.1~180=1),0,150000,#,7E4
004_Count_Speed,CountSpd,,val{004_Count_Speed_TMP},0,150000,#,7E4

Distance calculated from Real speed then will be:

004_CALC_Trip_Distance,DST,,(val{004_CALC_Average_Speed}*val{Trip time(whilst moving)})/3600,0.0,1000,km,7E4

migle, 2018-05-10:
There's also Torque's own trip distance. Is real speed more accurate than GPS? It should be more precise, but it looks a bit too low.

JejuSoul, 2018-05-11:
IoniqBEV - thanks for these calcs. I will test them for a week or so before adding them.
All our cars should be able to use them.

migle - there are many advantages to using the real vehicle speed PID rather than GPS. eg.
1/ In a tunnel you lose GPS signal.
2/ In a national park you lose GPS signal. I just saw a Bolt EV video where the driver lost all navigation in Zion national park.
3/ If you use an old tablet for Torque so that you can leave it in the car without worry. It doesn't have GPS at all.

A couple of things I would rewrite.
1/ keep all values in miles and mph. Allow Torque to do the conversions when it displays.
2/ LOOKUP requires the latest version of Torque. I prefer to add code that allows older versions to work as well. If there is a simple alternative I will change it. Otherwise we can keep it, but it needs to be commented that it only works with the latest version.

IoniqBEV, 2018-05-11:
There is still one problem: value of Trip time(whilst moving)- it is built in PID, time starts running if Torque detect motion through GPS signal...  without GPS will be calculated distance zero. I will try to make timer independent of GPS.
Real speed Torque automatically convert from mph to km/h, but calculated average speed remained in mph, therefore is multiplication 1.6 if I want to km/h

IoniqBEV, 2018-05-11:
migle: .." There's also Torque's own trip distance."..
 Where? Whitch PID?  I do not see any value of distance in Torgue

migle, 2018-05-11:
> migle - there are many advantages to using the real vehicle speed PID rather than GPS. eg.

Of course. I keep seeing only the perspective of accuracy... Some people may find that one useful.

> 2/ In a national park you lose GPS signal. I just saw a Bolt EV video...

Thanks for the tip. Nice video for a bit of youtube tourism. He didn't have any cells for communications. We are so used to the accuracy and speed of A-GPS that we forget that for true GPS signal we must keep the phone near the windshield, with lots of clearance (like any good old GPS device).

> I do not see any value of distance in Torgue

I don't know, maybe you need to create a profile first? See attached.

[image]

IoniqBEV, 2018-05-11:
This PID- Trip Distance- I see, but always stay zero. Is Your value changing?