JejuSoul/OBD-PIDs-for-HKMC-EVs #7: Ioniq EV lacks a lot of data points

kinnison, 2018-04-18:
Hi,

The Ioniq EV entry in this repo lacks a lot of information about the car.  In an attempt to make it possible to work things out, I have done a basic test set in my Ioniq and recorded the CAN traces.

My CAN probe program takes a sample every 50ms from 11 different queryable points (ECU/query) so there's a lot of data, but it should be easy to filter.

I did a basic drive, from the hackspace where I soldered my probe to the charger, and then a charge at a stand which seems to only ever offer me 15a, a charge at a stand which should do more (since it does 43kw AC) and then CCS.

I also did a standing-test where I toggled things like drive mode, VESS, etc.  A rough plan is listed in the zip file, and each CAN log is in candump format.

If you need more information, please ask.  I'm happy to do a variety of tests in the car.

[logs-for-jejusoul.zip](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/1923376/logs-for-jejusoul.zip)

D.

migle, 2018-04-18:
Can you tell me what does your global odometer read, more or less? I would really like to be logging odometer data...

kinnison, 2018-04-18:
The information is in the zip file, including odo reading, external temperature, etc.

kinnison, 2018-04-18:
The tests were done in the order of the test log file, so the drive from the hack space to the charger was before the odo reading in the file.

kinnison, 2018-04-18:
It may be split across packets remember.  Also there's no guarantee I'm querying the right ECU/message combo for getting it yet.

migle, 2018-04-18:
I don't know anything about CAN bus. Are you getting the entire frame? I understand that the data we see in Torque comes mostly from "extended" frames.

For that reason, it's frequent that cheap ELM-327 clones don't read a big part of the data. Say, they can read "BMS SoC" but they read garbage in "Display SoC". Often the real ELM-327 gets a bit confused too (I have one).

kinnison, 2018-04-18:
The CAN trace shows the full frame set.  OBD-II and by extension, the car-specific PIDs we're querying, is built on UDS (unified diagnostics) which is built atop ISO-TP atop CAN.  My probe supports ISO-TP and so the full ISO-TP *packets* are available in the trace, they simply span multiple CAN *frames*.

If you look you'll see frames which start 1 followed by one starting 3 followed by a number of 21, 22, 23, etc frames.  This is the ISO-TP on top of CAN.  the frames starting 0 are 'stand alone' where 1 is "here's the start of a long packet" 3 is "flow-control, send me more please" and 2x are the continuation frames in the packet.

If we look near the top of the basic drive log, we see:

```
7E2#0221010000000000
7EA#10166101FFE00000
7E2#3010000000000000
7EA#21092180A505E402
7EA#2200000000776330
7EA#2304200000000000
```

1. The first frame there `02 21 01 ...` is the request to 7E2
2. The second frame is the 'first frame' response saying there are 0x016 bytes in the packet (24)
3. The third frame is the flow control saying "Send me up to 16 more frames please"
4. The fourth, fifth, and sixth frames are the continuation of the response.

If we decode this, we get the full ISO-TP *packet* for the response as being: `61 01 FF E0 00 00 09 21 80 A5 05 E4 02 00 00 00 00 77 63 30 04 20` with the final zeroes in the last frame being padding.

UDS tells us that the `61 01` at the start is a positive response to the `21 01` request.  All of Hyundai's ECUs seem to fill the rest of the first frame wth some constant values (`FF E0 00 00` in this case) and then the rest tends to be decodeable as data.  The `FF` is, I think, value `a`, the `E0` is `b` etc. in Torque's equations.

I hope that helps you understand the log files a little more.

migle, 2018-04-18:
Ok. Thanks for the tutorial. So, I don't really gain anything by looking at the CAN frames. I'd better just use obd terminal and just ask 21 01 and read the response...

So, maybe in this dialog (this is from your CCS charge, where I could look for the current value which is kept constant at -125A):
`(1523991329.589631) slcan0 7E4#0221010000000000
(1523991329.605233) slcan0 7EC#103D6101FFFFFFFF
(1523991329.605782) slcan0 7E4#3010000000000000
(1523991329.618291) slcan0 7EC#213E26482648C3FB <- FB is K in Torque
(1523991329.628476) slcan0 7EC#222C0DCC0E0E0D0D <- 2C is L, 0DCC are M and N
(1523991329.638360) slcan0 7EC#230D0D0D000FB82C
(1523991329.648413) slcan0 7EC#24B7270000910000
(1523991329.658270) slcan0 7EC#259D5B00009DFE00
(1523991329.670142) slcan0 7EC#260039E200003784
(1523991329.678149) slcan0 7EC#270013251B090161
(1523991329.688333) slcan0 7EC#280000000003E800`

So, FB2C could be the current in 0.1A units, -123.6A, and 0DCC could be the voltage in 0.1V units (353.2V), if you actually were on such low state of charge. Actually, if this is true, SoC BMS would be byte e, so 3E in 0.5% units, so you would be at 31.0% SoC BMS.

kinnison, 2018-04-18:
Yep, my charge level was around that (to my memory) so those numbers all match up with what we expect from the BMS entries in this repository :-D

migle, 2018-04-18:
You must be careful. There's a lot of garbage there, as frequently happens with ELM-327 and Torque as well. Sometimes you ask 21 01 and get B2B2B2B2... as answer (or B4B4B4... or B6B6B6...).

Also, when reading your voltage (I was looking for a number that was rising on lines with 7EC#22), correct voltage values alternate with D0D (fgrep 7EC#22 charger-CCS.log | less). We also sometimes see correct values in Torque alternating with constant garbage values.

kinnison, 2018-04-18:
Different endpoint.  All those B4 and B6 responses are for the individual cell voltages.

kinnison, 2018-04-18:
Can you give me an example, I would have thought that those were the different requests for the cell voltages.

migle, 2018-04-18:
You're right. I was reading fgrep 7EC#22 and stupidly assumed that I was always reading the answer to 2101 which I was not.

migle, 2018-04-18:
> If you're comfortable with Linux,

I am.

> and have a CAN interface for your Linux system,

I don't. Do I? I've no idea.

But look: I'm not going below the ELM-327. On top of that, for asking something and getting the answer, I have my own program, if needed. (Because Torque is not very clever in that it asks all PIDs with the same frequency, there's no point asking for a temperature as often as you ask for RPM).

Is there a strategy to find out what to ask? I tried Torque Scan on a trip while my wife was driving. It took a lot of time, then it didn't save the results...

kinnison, 2018-04-18:
At one point, a chap on the German forum which sparked all this had done a scan and identified the ECU endpoints which responded to 2101 requests etc.  I'm afraid I'm not very well versed in the "how to discover stuff in the car" bit, because normally I'm working on an ECU, not a scanner ;-)

I bypassed the ELM327 and went straight to a USB-attached CAN interface because the ELMs are so slow and the bluetooth ones at least are quite lossy.

The input to my program is at https://git.gitano.org.uk/canopied.git/tree/definitions/raw.yaml and from there you can see the request packet (sendto address and packet contents) of what I'm aware of so far.  That's all raw stuff of course, I've not written up the conversions to real values yet.  Those for the BMS are in this repo already though.

migle, 2018-04-18:
> because normally I'm working on an ECU

You work on an ECU?? You're well placed for this, then! I'll try another round of scanning the car, or checking your findings soon.

JejuSoul, 2018-04-19:
Wow - that is a lot of data you have recorded.
I will have a proper look at this over the weekend when I have time.
After a brief glance from the first few lines I see that the BMS codes are as expected.

`7E4#0221010000000000
7EC#103D6101FFFFFFFF
7EC#213E264826480300
7EC#22080D6E0D0D0D0D
7EC#230D0D0D000DB326
7EC#24B2270000920000
7EC#259D5600009DF800
7EC#260039E100003782
7EC#2700131FEE0D0157
7EC#280000000003E800

7E4#0221050000000000
7EC#102D6105FFFFFFFF
7EC#2100000000000D0D
7EC#220D0D0D0D0D2648
7EC#2326480001550D0D
7EC#2403E85003E8273F
7EC#2500290000000000
7EC#2600000000000000`

What is new to me for the Ioniq EV is what I hope are the OBC codes at 7E5
I've picked out a couple of data points from the charger-32A-capable.log

`
7E5#0221010000000000
7ED#102C6101FFFFF800
7ED#2100793F0293040B
7ED#22001377003401BD
7ED#2305B04100000000
7ED#2400040000680017
7ED#25000D01F4867B03
7ED#263B001200000000
`
and 

`
7E5#0221010000000000
7ED#102C6101FFFFF800
7ED#210192FD871060E6
7ED#2207747B0245006E
7ED#2305A443102000C8
7ED#240AD1037776001A
7ED#25000C01F4879584
7ED#26A9001400000000
`

kinnison, 2018-04-19:
Yeah, I'm both sorry and not-sorry that the logs are quite verbose :-)  I'm glad that the BMS looks sane to you :-)

If you want me to do any tests with specific cases, let me know.  I'll be able to make recordings of charge sessions etc, add requests, change cadences, etc as you require.

migle, 2018-04-19:
Using cut, paste and friends, I merged the ISO-TL packets and got rid of the CAN frames, separated by request type and endpoint, sorted by timestamp. With one response packet per line and only responses of the same kind in one file, it's easier (for me) to spot patterns.

[user-friendly-logs.zip](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/1930054/user-friendly-logs.zip)

I don't think the odometer is there, the one I would like to find. If it were there should be numbers changing monotonically, some faster, some slower, and I don't see that. One suspect would be octet 21 in the response of 7D5/7DD to 2101. It goes slowly from C4 to C7 and could form a team with any of the octets around. But I don't think it is, I would expect the entire number to be there in sequence, many non-zero digits.

JejuSoul, 2018-04-20:
Hi migle,
I didn't expect to find the odometer in these requests.
I am assuming much of the coding is similar to the Kia Soul EV
We can see that data analysed here - https://docs.google.com/spreadsheets/d/1YYlZ-IcTQlz-LzaYkHO-7a4SFM8QYs2BGNXiSU5_EwI/edit?pref=2&pli=1#gid=0
The odometer value is on the C_Can in 24bit Little Endian format.  LE (km * 10)

kinnison - your BMS data is good. but I don't think it is the OBC at 7E5 -> 7ED
Am assuming that most values from the OBC will be zero when the car is not charging. 

State of Charge of Battery(BMS)	31.00	%
Available Charge Power	98.00	kW
Available Discharge Power	98.00	kW
Battery Current	8.00	A
Battery DC Voltage	343.80	V
Battery Max Temperature	13.00	°C
Battery Min Temperature	13.00	°C
Battery Module 01 Temperature	13.00	°C
Battery Module 02 Temperature	13.00	°C
Battery Module 03 Temperature	13.00	°C
Battery Module 04 Temperature	13.00	°C
Battery Module 05 Temperature	13.00	°C
Battery Inlet Temperature	13.00	°C
Max Cell Voltage	3.58	V
Max Cell Voltage No.	38	
Min Cell Voltage	3.56	V
Min Cell Voltage No.	39	
Auxiliary Battery Voltage	14.60	V
Cumulative Charge Current 	4027.80	A
Cumulative Discharge Current	4044.00	A
Cumulative Charge Energy	1481.70	kWh
Cumulative Discharge Energy	1421.00	kWh
Cumulative Operating Time	348.16	hours
Inverter Capacitor Voltage	34.30	V
Isolation Resistance	1000.00	kOhm
Battery Module 06 Temperature	13.00	°C
Battery Module 07 Temperature	13.00	°C
Battery Module 08 Temperature	13.00	°C
Battery Module 09 Temperature	13.00	°C
Battery Module 10 Temperature	13.00	°C
Battery Module 11 Temperature	13.00	°C
Battery Module 12 Temperature	13.00	°C
Available Charge Power	98.00	kW
Available Discharge Power	98.00	kW
Airbag H/wire Duty	85.00	%
Battery Heater 1 Temperature	13.00	°C
Battery Heater 2 Temperature	13.00	°C
State of Health SOH	100.00	%
Max Deterioration Cell No	80	
Min Deterioration	100.00	%
Min Deterioration Cell No	39	
SOC Display	31.50	%

kinnison, 2018-04-20:
Nice to have the BMS confirmed.  Okay, so the extra thing is likely not the OBC.  Fair enough.  The VMCU decodes which were previously provided do seem sane (brake pedal switch, brake light indicator, P/D/R/N indicator at least).

So, currently I'm requesting:

```
 BMS - 7E4 - 2101 2102 2103 2104 2105
VMCU - 7E2 - 2101 2102
???? 7E5 - 2101
???? 7E3 - 2101
???? 7D5 - 2101
```

We thought `7E5` was the OBC mostly because it responds to an `090A` with `BCCM-B+ChargerCtrl` as its name (similarly to how `7E4` responds `BECM-B+EnergyCtrl`

None of `7E2`, `7E3`, or `7D5` respond usefully to `090A`.

I remember there was a non-trivial number more of CAN IDs listed which responded to 2101, perhaps I should try doing a sweep/scan?

I am attempting to book my Ioniq in to the dealer for a firmware update (I was told there should be one this month) and when I do, I'll ask if they have any printouts they could provide of ECU checks.

In the meantime I guess we're poking about trying to find patterns in what we already know about.

migle, 2018-04-20:
> The odometer value is on the C_Can in 24bit Little Endian format. LE (km * 10)

How does one query those values over the OBD port? Or it's not available through that?

kinnison, 2018-04-20:
I'm afraid that whatever CAN is on the OBD-II connector on the Ioniq is *silent* except when queries are sent.  There is no background chatter like you describe there :/

kinnison, 2018-04-20:
Unless there's commands I should be sending to turn it on which I just don't know, of course :-D

JejuSoul, 2018-04-20:
I looked up what the acronym BCCM means when talking about an EV:
Two possibilities (these may be the same thing)
 'Battery Charger Control Module ' used by Ford
 'Battery Charger Converter Module ' used by Magna Steyr

Looking here - http://www.freepatentsonline.com/y2017/0028857.html for the Ford meaning
and here - http://www.energy.gov/sites/prod/files/2014/03/f13/arravt027_ape_peaslee_2013_p.pdf for the Magna meaning
Neither of these is an OBC.

kinnison, 2018-04-20:
If you have a Soul, are you able to issue 090A to the OBC and see what it responds?
Even if it's not a full name it returns, we might be able to match the kind of error code it generates to one of the other EMUs in the Ioniq.

JejuSoul, 2018-04-22:
>If you have a Soul, are you able to issue 090A to the OBC and see what it responds?

I tested all the known ECUs from the Soul EV.
Only the BMS gives a useful response - BECM-B+ChargerCtrl
The VMCU does respond in the same way - but the 23 bytes are all zero.

The OBC response I assume is an error code - 03 7F 09 11

All the other ECUs did not respond.

kinnison, 2018-04-22:
> The OBC response I assume is an error code - 03 7F 09 11

Yeah, that's an error code.  Oh well, it was worth a go.  So we can guess that an ECU which returns an explicit error code might be the OBC.  I shall have a go at finding any such ECUs as soon as I can (likely tomorrow)

JejuSoul, 2018-04-22:
There's a typo in my comment above. On the Soul EV the BMS responds to  090A with  BECM-B+EnergyCtrl. 

Could you try logging a couple of different ECUs. Finding the OBC and LDC really should be fairly easy once you can spot a temperature variable. The temperature of these 3 parts rise and fall in sync during driving or AC charging. The temps are static when parked or DC charging.

I've had another look at the responses from the 7E5 requests, and still have no idea what they are.

I'll look next at the VMCU codes.  Try and see why ITStromer deleted his temperature data.

kinnison, 2018-04-23:
I will see what I can discover this evening.  If I can find an ECU responding to 090A with 03 7F 09 11 then we might have a candidate for the OBC.

kinnison, 2018-04-23:
Several ECUs respond that way sadly.  None of them seem to produce useful stuff on 2100/2101/2102/2103 excepting 7e5 which we've discounted, 7e3 which you didn't hilight before but seems to respond to 2101 2102 and 2103 (I've added 2102 and 2103 to my tester app and will post a new log) and 7d5 which you didn't hilight before either.

I'll get an AC charger log with 7e3's 2102 and 2103 results too and we can look at those.

kinnison, 2018-04-23:
One other thing, the brake switch vs. brake lamp always seem to be opposites.  Even if the brake lamp is on because of auto-hold, the value is false unless the brake pedal is pressed.

kinnison, 2018-04-23:
I have taken some new logs with the extra PIDs included.

[logs-for-jejusoul2.zip](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/1939733/logs-for-jejusoul2.zip)

Sorry it's quite so huge, but it's 41ish minutes long covering a charge session from 89% to 100% using an AC charger.  If any of these PIDs are involved, hopefully you can spot them this time.

If you think you've found anything, let me know and I'll add the equations and re-play the log to see

kinnison, 2018-04-23:
From a brief look, I'd guess that the new PIDs don't help much, the 2102 doesn't seem to respond (or if it does, it's super slow) and the 2103 seems to always be zeroes.

JejuSoul, 2018-04-25:
>    Neither of these is an OBC. - For a layman, it sounds the same.
Have been thinking more about whether the BCCM could be the same as the OBC.
I said no earlier because the BCCM is something to do with opening the DC connectors for fast charging.
The OBC is only for AC slower charging.

But the OBC in the Ioniq is  integrated with the DC-DC converter . And this may be the BCCM.
The Soul EV has an OBC by a Korean company Mando, whereas the Ioniq has an OBC by another Korean company Mobis.

I've written about this here - http://www.mykiasoulev.com/forum/viewtopic.php?t=455&start=110#p5062

There's a newer version that supports V2G -  http://www.greencarcongress.com/2017/08/20170816-mobis.html

JejuSoul, 2018-04-25:
I really wish we had the PID code to pull the ROM ID from each ECU.
see my comment here - http://myioniq.com/forum/viewtopic.php?f=8&t=73&p=484#p484

[img]http://oi64.tinypic.com/2v14654.jpg[/img]

JejuSoul, 2018-04-27:
>One other thing, the brake switch vs. brake lamp always seem to be opposites. Even if the brake lamp is on because of auto-hold, the value is false unless the brake pedal is pressed.

Can you explain the auto-hold feature.

The brake switch seems to be the opposite of a common sense value. 1 = brake not pressed, 0 = pressed.
I think we should inverse this value, but Torque lacks the bitwise NOT operator !
Maybe I'll use the equation {J:1} * -1 +1
Anyone got any thoughts on this.

Perhaps the brake lamp PID should just be renamed brake pedal pushed.
Does the PID goto 1 when the brake lamp comes on due to regen?

JejuSoul, 2018-04-29:
I just did a test of the renamed  'Brakes On' PID on my Soul EV. 30 minutes driving in B mode and 30 minutes in D mode. (equivalent to regen 2 and regen 1 on the Ioniq)
In B mode the brake lamp is on when the brakes on switch is on and vice versa.
But
In D mode there were 8 occurrences of a few seconds each where
The car is doing regen but not slowing down, the brake lamp is off and the brakes on switch is on.

ITStromer, 2018-05-09:
> So, currently I'm requesting:
>BMS - 7E4 - 2101 2102 2103 2104 2105
>VMCU - 7E2 - 2101 2102
>???? - 7E5 - 2101
>???? - 7E3 - 2101
>???? - 7D5 - 2101

7D5 seem not to be existent in my german version of the Ioniq EV. Could it be part of bluelink? I thought only the american versions have one, right?

But additionally we have another ECU:
7E6 responds to 2180 and 2181

We are working on my google sheet to find more PIDs.
Who is interested can have a look on it and may comment: https://docs.google.com/spreadsheets/d/1R2dW_4-ANg04TQdkHH52-8wdYItBkWDfppcsqOc-m1M/edit?usp=sharing

If you want to contribute directly into the document, contact me.

kinnison, 2018-05-10:
I'll try adding 7E6/2180/2181 to my program and see if I can get anything useful out of it.

If you want any particular recordings to improve your research @ITStromer then let me know.

I don't have bluelink (UK version of the car), so I have no idea what 7D5 might be.  I have the "top trim" for the UK, (Premium SE) which lacks things like a sunroof (dunno why I don't get that :-)

asmaps, 2019-03-02:
any news on how to get the ODO value from Ioniq BEV?
I programmed my own app, so I'm maybe a little more flexible but going into listening mode normally doesn't work well with the cheap dongles. Anyone found a PID/Command to issue to get the odo?

migle, 2019-03-02:
No, not me. I think there are other things more easily discoverable, such as the OBC. My hope is low for the odometer...

Listening mode has been pointless with my non-cheap ELM-327 and my non-cheap OBDLink LX. Other users here were using generic CAN devices, so unless you find another plug in the car, listening is pointless.

ITStromer, 2019-04-25:
I found the outside temp sensor at ECU 7e6!
But I'm unsure how to name the ECU. There are more temperature sensors. One seems to be the VMCU heat sink another maybe the heating element temp sensor.

In my fork I included the outside temp sensor into Hyundai_Ioniq_EV_VMCU_data.csv like this:
003_VMCU outside temp	outside temp	2180	(m-80)/2)	-40	80	C	7E6

What do you think, should I pull it so back into JeJuSouls main branch or should we use another file? I would prefer to not open too much file for so few values - maybe later if we find more.

joaand, 2020-06-13:
I added a file for Odometer in my fork.

004_Odometer,Odo,22b002,INT24(G:H:I),0,16777215,km,7c6

migle, 2020-06-15:
Thanks! What a breakthrough! The odometer was the most important information missing.

JejuSoul, 2020-06-15:
Great. Thanks for posting this new PID.
Can someone else confirm it works.

One technical issue. I will probably adapt your code to use an older version of INT24(G:H:I)
This is a new function and the only mention in the November 2019 changelog states - 
 *  Added INT24(A:B:C) function (in next beta update)
see https://torque-bhp.com/forums/?wpforumaction=viewtopic&t=3.0

I prefer to keep the code usable for very old copies of Torque. 
I like to leave an old phone in the car to record charging.
The software on this phone does not get updated.

joaand, 2020-06-15:
It works this way too:
(G*256*256)+(H*256)+I

JejuSoul, 2020-06-16:
I began testing the new INT16, INT24, INT32 functions in Torque version 1.8.202
They work, and look so much neater and simpler to understand than the clunky workarounds we have always used. This version of Torque is now already a year old, I think many users will already have updated.

Also do you know which ECU is responding to 7C6 on the Ioniq EV. 
On my SoulEV that is the Cluster. It makes sense that the Odometer PID should be found on the Instrument Cluster. I will check to see if that is common across all Hyundai/Kia vehicles.

migle, 2020-06-16:
I have confirmed that the odometer pid works on the Kona 64kWh.

migle, 2020-06-16:
I would additionally like to comment, but I wouldn't open a new ticket for this, that I think the operating time would be more useful like this:

```
000_Operating Seconds,OpSecs,2101,(au<24)+(av<16)+(aw<8)+ax,0,1000000000,seconds,7E4
```

This is almost the same as the existing OpTime but it is an integer. As such, it is easy to log in a text file without loosing any precision.

In particular, because no other PID uses more than two decimal places, it allows people to use the option in Torque to format the logs files to two decimal places, making numbers more clear and saving a lot of bytes.
Operating hours with two decimal places becomes meaningless.

We now have a time and space (odometer) in the vehicle data!

JejuSoul, 2020-06-17:
Good. We have confirmation for this new odometer PID. 
I accept the change to OpTime.
Given the new functionality I would rewrite the code like this.
```
000_Operating Seconds,OpSecs,2101,INT32(au:av:aw:ax),0,1000000000,seconds,7E4
000_Operating Time,OpTime,2101,val{000_Operating Seconds}/3600,0,1000000,hours,7E4
```

sixtybi, 2021-01-23:
Hi,

On Ioniq EV 28 is that possible to read the state of some push button? Or the state of any subsystem?
Eg lane assistant state?
I would like to create a script which can open my garage door based on some trigger, eg switch off lane keep assistant (which I never do), or press a button which I don't use. Would this be possible by reading any PID?

joaand, 2021-01-24:
Use a geofence and a combination of the car being in N and the brake being pressed.

This should be a new issue, it has nothing to do with the existing thread.

sixtybi, 2021-01-28:
> Use a geofence and a combination of the car being in N and the brake being pressed.
> 
> This should be a new issue, it has nothing to do with the existing thread.

Thanks! Yes, I was thinking about this idea and I guess I will try something similar to this.
I would like to do something which doesn't influence the driving and can be somehow catched by the obd2. This gear to N seems OK, I was just testing this.
I guess I will check if gear is in N, while speed is more than 0 and location is nearby.
Thanks again for the hint!