JejuSoul/OBD-PIDs-for-HKMC-EVs #10: Same bytes used for different PIDs

ITStromer, 2018-05-09:
Is it correct, that the following PIDs use the same byte-values? I believe there is an error in one of them:
003_VMCU Aux Battery Voltage: 7E2, 2102, e/10
003_VMCU Inverter Input Voltage:	7E2, 2102, e*2

And here probably a second mismatch:
003_VMCU Heat Sink Temperature: 7E2, 2102, y-40
003_VMCU Motor U Phase Current Sensor Offset: 7E2, 2102, ((z<8)+y)

I found them in Kia_Soul_EV_VMCU_data.csv while searcing for Ioniq correspondences.

JejuSoul, 2018-05-10:
Thanks for posting this. They are obviously errors. I will have a look on my car to see which of them is correct. And then fix the other.
The data table they are derived from is here - https://docs.google.com/spreadsheets/d/1YYlZ-IcTQlz-LzaYkHO-7a4SFM8QYs2BGNXiSU5_EwI/edit?pref=2&pli=1#gid=1449971753

JejuSoul, 2018-05-10:
Am looking at making the following corrections in the Soul EV VMCU file.

003_VMCU Inverter Input Voltage: 7E2, 2102, f*2
003_VMCU EWP speed,EWP speed,2102,g,0,190,,7E2
003_VMCU EWP Control Target RPM Request,EWP Target,2102,h,0,190,,7E2

003_VMCU Motor U Phase Current Sensor Offset: 7E2, 2102, ((aa<8)+z),0,1000,,7E2
003_VMCU Motor V Phase Current Sensor Offset,V Phase,2102,((ac<8)+ab),0,1000,,7E2

will go and test them before adding to the repo.

JejuSoul, 2018-05-12:
There were errors in the codes listed on this page - corrected versions here - 

Have made the following corrections in the Soul EV VMCU file. Both 27kWh and 30kWh

003_VMCU Inverter Input Voltage,Inverter,2102,f*2,0,360,V,7E2
003_VMCU EWP speed,EWP speed,2102,g,0,190,,7E2
003_VMCU EWP Control Target RPM Request,EWP Target,2102,h,0,190,,7E2

003_VMCU Motor U Phase Current Sensor Offset,U Phase,2102,((aa<8)+z),0,1000,,7E2
003_VMCU Motor V Phase Current Sensor Offset,V Phase,2102,((ac<8)+ab),0,1000,,7E2

closed 2018-12-11 by ayokyun