JejuSoul/OBD-PIDs-for-HKMC-EVs #11: RPM for the Ioniq EV is wrong

JejuSoul, 2018-05-13:
ITStromer added this PID 
003_VMCU Motor Actual Speed RPM,Speed rpm,2102,((Signed(F)*256)+E),-10100,10100,rpm,7E2

but I think it is wrong.
I have looked at his spreadsheet and he seems to have found this value at 7E3 not 7E2

I want to change this value to 
003_VMCU Motor Actual Speed RPM,Speed rpm,2102,((Signed(F)*256)+E),-10100,10100,rpm,7E3

does everyone agree?

JejuSoul, 2018-05-14:
Okay - we have agreed this was an error I have fixed it.
But we now have to work out which ECU is 7E3
Or does one ECU the VMCU respond on two different codes 7E2 and 7E3

migle, 2018-05-14:
Either I made a mistake typing the formula, or it isn't right. See attachment. I'll check tomorrow.
Is there a simple way to log all response bytes in Torque? (Easier than adding a formula for each byte?) Because that would make testing hypotheses a lot less easier.

[trackLog-2018-Mai-14_08-16-32.txt](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/2002784/trackLog-2018-Mai-14_08-16-32.txt)
.