JejuSoul/OBD-PIDs-for-HKMC-EVs #12: ABRP Support

Felger, 2018-09-04:
We'd like to support the Ioniq EV (and any other Hyundai EVs you'd like us to support as well) on [A Better Route Planner](https://abetterrouteplanner.com), would you be willing for us to use your PID lists for people to contribute driving data so we can build accurate driving models for these vehicles?

Thanks in advance!

ayokyun, 2018-09-05:
As a one of the contributor, I think it would be great idea.

BTW, how can I add charger information to ABRT?

Felger, 2018-09-05:
Depends on where you are, it's on my to-do list to add links to add / edit the chargers.  The main databases we pull from (in order of priority) are:
http://uppladdning.nu/
http://goingelectric.de/
http://openchargemap.org/

In general, if you're in Europe, pick between Uppladdning.nu or Going Electric.  Everywhere else in the world, Open Charge Map. Of course, we're very open to adding other charger databases to improve our coverage, we don't want to be yet another database, instead leveraging the data that's already out there.

Once I build my instructions, would you be willing to review them for accuracy?

Felger, 2018-09-05:
Oh, and you can get a preview of our Ioniq and Kona models at:
https://abetterrouteplanner.com/beta/
We're looking for feedback to check accuracy before we release them to the world, if you know anyone who's got a Kona yet, send them our way!

Felger, 2018-09-06:
Until we get a lot of contributor data, we leave the models intentionally conservative so we don't over-promise performance.  How conservative is that? Having no real frame of reference, I built the model on what I could dig up on rolling resistance, aerodynamics coefficients, and weight.  

You can adjust the way planning works easily by entering your standard consumption efficiency in km/kWh or Wh/km (just select the units button next to the input field).

We'll have a look at those maps, we're pretty limited to those charger databases that allow public access (or cheap / free private access).  We're supported by donations, so we can't afford the exorbitant fees Plugshare and other large databases require.

JejuSoul, 2018-09-19:
> would you be willing for us to use your PID lists 
Yes.
>Do you have any intentions of adding a PID list for the Kona
Yes. Also the Niro EV.

closed 2018-12-11 by ayokyun