JejuSoul/OBD-PIDs-for-HKMC-EVs #13: Ioniq Parameters mostly have the same PID

Felger, 2018-09-05:
Working to set up the ABRP webserver to receive Torque Pro data using the Ioniq PID list, noticed that the PIDs nearly all have the same entry in the "Mode and PID" column, which Torque uses to identify that parameter to the webserver.  For example, the line:
```
000_Battery Current | Batt Current | 2101 | ((Signed(K)*256)+L)/10 | -230 | 230 | A | 7.00E+04
-- | -- | -- | -- | -- | -- | -- | --
```
Defines the PID for the battery output (input) current, which uses the PID 2101.  This is common in the rest of the PID list for BCM values.  Unfortunately, when Torque sends this data to the webserver, it uses 2101 as the identifier.  Example data from Torque webserver packets:
```
&defaultUnits2101=A
&k2101=10.7
```

migle, 2018-09-05:
We can't change the PIDs here... You have to report this issue on Hyundai's github... (joking) We can't change the car.

That's how it works. When we send an OBD request with 2101 we get a packet in response with many bytes. Two of those bytes (K and L) encode the battery current, others encode other things, voltage, whatever.

Felger, 2018-09-06:
I had an idea, and applied it to the PIDs I'd want to track to build a good Ioniq model, I think it should work.  I'll see if I can find someone with an Ioniq to test it and see.

Felger, 2018-09-07:
Resolved via #14

closed 2018-09-07 by Felger