JejuSoul/OBD-PIDs-for-HKMC-EVs #21: VMCU set on Ioniq HEV don't work

gabry79, 2018-10-05:
I tried all sets of PIDs and the BMS data is working very well. Also the BMS cell data.

But the VMCU set seems to be unuseful. I can't read any value. Can you help me?

JejuSoul, 2018-10-19:
I have a Kia EV.
Hyundai and Kia share the same BMS.
They do not share the same VMCU.
But there are many similarities.
For the Ioniq HEV I suggest trying codes already tested on the Ioniq Electric.

JejuSoul, 2018-10-19:
Also I suggest looking at this thread on the Ioniq forum - https://www.ioniqforum.com/forum/153-powertrain-technical-discussion/11009-ev-ice-engines-having-asynchroneous-rpm.html

gabry79, 2018-10-20:
Thank you very much for the answer.

I already tried the set of VMCU PID for Ioniq EV, but it doesn't still work.

I need only few parameters that now can't be read: instant kW power/Nm torque of ICE, instant fuel consumption (possibly real injected gasoline), know when we are really braking with brakes (instead of regenerative charging), the effective gear of DCT.

In your opinion, since you are an expert, could we be able to get this data through a terminal connected to the OBD?

lysiong, 2020-06-24:
I also would like to know the PID value for current DCT gear position.