JejuSoul/OBD-PIDs-for-HKMC-EVs #22: About Kona EV ?

asetyde, 2018-10-10:
We are working on external monitor system auto battery ... but who help us say about ionic commands are not compatible with Kona ! 

Do you can support it ? Kona will increase abut market more thanks ionic soon

asetyde, 2018-10-10:
Hi,
I'm embedded engineer with Hyundai Kona EV, I see your project but I want do a similar but more simple system to monitor charge of vehicle.
I want use this scheme :
OBD2 -> EMBEDDED SIM DEVICE -> SERVER/APP
OBD2-SIM DEVICE -> SMS/SERVER SERVICE
OBD2 with interface to embedded device esim -> to app server or direct to sms or similar (integrate with EVNotify ? )

But at first I need to understand how read from obd and what , can you help me ? I have iOS but also android in my lab

asetyde, 2018-10-10:
https://freematics.com/store/index.php?route=product/product&path=20&product_id=82

asetyde, 2018-10-10:
we can start from here and i find a support esim device
http://arduinodev.com/hardware/obd-kit/

JejuSoul, 2018-10-19:
Great. Lets us know what you find.

ayokyun, 2018-10-26:
@asetyde  Did you check freematics OBD is working with Kona EV?
I also have freematics ONE device, but it cannot communicate to some vehicle models, especially EV models.

ayokyun, 2018-10-31:
I've added some PIDs for Niro EV and am going to update more.
Would you check if it's working on Kona EV?

asetyde, 2018-11-06:
I'm waiting freematics and after doing test !!

plord12, 2018-12-05:
I tried the Niro csv files in Torque Pro on my Kona Electric and I do get a lot of sensible data :-)

asetyde, 2018-12-05:
@plord12  you use freematics ??

asetyde, 2018-12-05:
i have but I not start yet .. you can help me ? how info ? please tell us :)

plord12, 2018-12-05:
> @plord12 you use freematics ??

No.  I have Torque Pro and trying to get autopi working.

plord12, 2018-12-05:
> i have but I not start yet on my Kona EV .. you can help me ? how info ? please tell us :)

I think the idea is to look at the Niro cvs files and try and convert them to whatever freematics uses.

ayokyun, 2018-12-05:
Unfortunately Freematics cannot communicate HKMC vehicles.
I've tested Freematics with Niro but it doesn't work.
Many people complain compatibility issue of Freematics to electric
vehicles, but he doesn't think it's his issue.

asetyde, 2018-12-05:
oh dear ! Any other idea ?

plord12, 2018-12-07:
Sorry to hear that .. there was some discussion on the autopi forums about the assumptions of ICE cars.  Good news is that autopi now seems to work better with electric cars, although there is still some work to be done.

asetyde, 2018-12-10:
Autopi is more expensive, how use linux to do this simple work ?

asetyde, 2018-12-10:
But I ve order an autopi 4g , therefore when will arrive I can collaborate with you plord, Kona community in Italy , but I think all world are waiting for a solution

plord12, 2018-12-10:
autopi + kona are being discussed at least at -

* https://www.speakev.com/forums/hyundai-kona.281/
* https://www.speakev.com/threads/kona-autopi.130284/
* https://community.autopi.io/
* https://community.autopi.io/t/hyundai-kona-ionic-ev/723

So, yes, I'm hopeful those interested can collaborate and make make progress :-)

asetyde, 2018-12-17:
Ok now I ve autopi, at now , to understand situation , we need widget for autopi cloud panel ? how can we help ? 
Also I want use autopi as hotspot wifi for Kona nav system, therefore I can set with terminal ? 
i want do some work Saturday .. you have some advice or idea ?

plord12, 2018-12-17:
Well, pending auto pi being fully ready to accept and process custom PID's, I would say trying to figure out the Kona PID's using torque pro or other tools is useful - that way, when autopi is ready, we can slot in the info.

Personally I'm a bit stuck since we can't listen to all the CAN traffic - one hope is to get a 6pin to obd2 cable so that we can use the diagnostics port in the engine compartment.

asetyde, 2018-12-18:
" pending auto pi being fully ready to accept and process custom PID's"   it's not operative at now ?

Niponyoshi, 2018-12-19:
Hi. 
The update for custom’s widget and pids for autopi.io will come in a few day. Before 2019. 
So very very soon. 
I have autopi.io, and wait for it. 
To wait I make an iPhone app to show SoC of the kona.

asetyde, 2018-12-21:
Wow it can be a beautiful gift :)

alklnofx, 2019-02-22:
On the OBD plug of the Kia Niro EV it's the same, no CAN traffic on PIN 6 & 14 as in Kia Soul EV :-(
Only OBD requests are possible. 
Has somebody got acces to the Kona or Niro C_CAN bus anywhere else?

asetyde, 2019-02-22:
EV Notify show data of Kona

asetyde, 2019-02-22:
[image]

asetyde, 2019-02-22:
[image]

gtrdriver, 2019-03-21:
Im Working on a Solution with a Raspberry PI + Cable/USB OBD Adapter in kombination with a Kona EV and the Pid´s from this Site.

Im Testing my Solution now since 10 Days it works like a charm

asetyde, 2019-03-21:
@gtrdriver  use Autopi , we use with success .. it's complete and with cloud

asetyde, 2019-03-21:
we 've collaborate many people and we 've all data ! 
[image]
[image]

orcinus, 2019-03-23:
Waiting for my autopi, and in the meantime, trying to get Navdy's OBD to work with Kona EV.
Does anyone know whether Kona outputs the speed PID, and if yes, what the speed PID is?

It seems like it isn't returning anything on the usual PID 13.

plord12, 2019-03-23:
> Waiting for my autopi, and in the meantime, trying to get Navdy's OBD to work with Kona EV.
> Does anyone know whether Kona outputs the speed PID, and if yes, what the speed PID is?
> 
> It seems like it isn't returning anything on the usual PID 13.

I've been trying to find this as well !  I came across this GDS screenshot -

[image]

So I'm wondering if its pid 220100 "ad"

gtrdriver, 2019-03-23:
Is there anyone who get Indoor and outdoor Temperature ?

On GDS i get both Values ...

plord12, 2019-03-23:
> Is there anyone who get Indoor and outdoor Temperature ?
> 
> On GDS i get both Values ...

On the Kona, yes.

[image]

But I notice sometimes I get some out-of-spec data ( might be autopi ).

plord12, 2019-03-23:
> On GDS i get both Values ...

You've got access to a GDS ?  If so, it would be really useful to grab some screen shots and share.

gtrdriver, 2019-03-23:
Hi

What PID´s you use for Temp (inside and outside) on KONA EV ?

Yes - i have access to GDS - you if you need screenshots please contact me what data you need !

plord12, 2019-03-23:
> Hi
> 
> What PID´s you use for Temp (inside and outside) on KONA EV ?

I used the data here - https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/blob/master/Hyundai%20Kona%20EV%20%26%20Kia%20Niro%20EV/extendedpids/002_Kona%26Niro_EV_HVAC.csv - so 220101 / 7B3 / f and g

I already shared the autopi as a library BTW.

> Yes - i have access to GDS - you if you need screenshots please contact me what data you need !

Woo ... basically I would say every sensor screen !  This gives us some clues to decoding the responses.  Right now I'm trying to find speed, odometer and tyre pressures.  

I don't suppose the GDS has a way to show the PID's does it ?

Cheers.

gtrdriver, 2019-03-23:
Hi

I dont unerstand it

https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/blob/master/Hyundai%20Kona%20EV%20%26%20Kia%20Niro%20EV/extendedpids/002_Kona%26Niro_EV_HVAC.csv

Here is PID 220100 shown wich deliver only zero data

You wrote 220101 / 7B3 

So whats right here 220100 or 220101

Next Monday i can make Screenshots of all GDS Sensor data.

plord12, 2019-03-23:
> You wrote 220101 / 7B3
> 
> So whats right here 220100 or 220101

Oops, sorry, typo.  220100.  Here is my autopi config -

[image]

And here is a hex dump from my Kona  -

```
pid 220100 @ 2019-03-23T10:50:56.868873
7BB 10 26 62 01 00 7E 50 07
7BB 21 C8 FF 7C 67 66 0E E5 [?=200] [?=255] [INDOORTEMP=22] [OUTDOORTEMP=11.5] [EVAPORATORTEMP?=11] [?=14] [?=229]
7BB 22 8F FF FF 8F FF 11 FF [?=143] [?=255] [?=255] [?=143] [?=255] [?=17] [?=255]
7BB 23 FF FF FF FF FF FF FF [?=255] [?=255] [?=255] [?=255] [?=255] [?=255] [?=255]
7BB 24 FF 3A D6 8F 90 43 FF [?=255] [?=58] [?=214] [?=143] [?=144] [speed?=41.6407706650093] [?=255]
7BB 25 FF 00 FF FF 00 00 00 [?=255] [?=0] [?=255] [?=255] [?=0] [?=0] [?=0]
```

> Next Monday i can make Screenshots of all GDS Sensor data.

Brilliant !  Many thanks.

gtrdriver, 2019-03-23:
Hi

first of all - i dont use a AutoPI - the Reason: this is only a personal Projekt and i dont want to spend 300 Euro for this ...

Im using a Raspberry with a common OBD Adapter (cable based) a 3G Internet Stick and a Common USB  GPS REciever. i rote a small python script wich takes the Hex Data out of the Adapter decode it and write it to a MYSQL Database on a Internet Server wich i own for other things ...

As i told i get all pid data and can decode nearly nearly every pid from this page - but on pid 220100 i only geT:

```
`220100
7ED 03 7F 22 12
7EA 03 7F 22 12
7EB 03 7F 22 12
```
`And i think you will agree there is no temp information.....

I have tryed 7 OBDAdapters here  - one with a original ELM Chip - but allways i get the same REsult ...

Regarding your Question GDS
No - GDS dont display the PID Number ... - but if it helps i will post some DATA SCreenshots of a Kona EV 64 with all Pid data GDS will deliver ...

plord12, 2019-03-23:
I believe ` 7F 22 12` is an error response.

Do you have torque pro ? Maybe try than as a comparison with the cvs files here.

But I'm afraid I don't know enough about the protocol to help much ... ( although having said that, maybe I could enable trace mode on my autopi - this gives some low-level info that might help ).

plord12, 2019-03-23:
Oh, I already have a debug trace -

```
2019-03-10 14:48:35,663 [salt.loaded.ext.engines.obd_manager:65  ][DEBUG   ][822] Querying: Outdoor_Temperature
2019-03-10 14:48:35,670 [i2c_conn         :104 ][DEBUG   ][823] Read byte 2/6 in block: 11111111
2019-03-10 14:48:35,677 [obd.obd          :267 ][DEBUG   ][822] Querying command: 220100: None
2019-03-10 14:48:35,674 [salt.loaded.ext.engines.obd_manager:614 ][DEBUG   ][822] Listener got battery result: {'_type': 'bat', 'voltage': 14.9, 'state': 'charging', 'level': 100}
2019-03-10 14:48:35,690 [i2c_conn         :104 ][DEBUG   ][823] Read byte 3/6 in block: 11000000
2019-03-10 14:48:35,687 [obd.interfaces.elm327:827 ][DEBUG   ][822] Write: 'ATSH7B3\r'
2019-03-10 14:48:35,699 [obd.interfaces.elm327:883 ][DEBUG   ][822] Read: b'OK\r\r>'
2019-03-10 14:48:35,704 [i2c_conn         :104 ][DEBUG   ][823] Read byte 4/6 in block: 11101111
2019-03-10 14:48:35,710 [obd.interfaces.elm327:509 ][DEBUG   ][822] Changed header from '7DF' to '7B3'
2019-03-10 14:48:35,719 [i2c_conn         :104 ][DEBUG   ][823] Read byte 5/6 in block: 11110000
2019-03-10 14:48:35,715 [obd.interfaces.elm327:827 ][DEBUG   ][822] Write: '220100\r'
2019-03-10 14:48:35,728 [mma8x5x_conn     :317 ][DEBUG   ][823] Calculating G for block: 0000000000000010
2019-03-10 14:48:35,733 [mma8x5x_conn     :317 ][DEBUG   ][823] Calculating G for block: 0000001111111111
2019-03-10 14:48:35,741 [mma8x5x_conn     :317 ][DEBUG   ][823] Calculating G for block: 0000001110111111
2019-03-10 14:48:35,831 [obd.interfaces.elm327:883 ][DEBUG   ][822] Read: b'7BB 10 26 62 01 00 7E 50 07 \r7BB 21 C8 FF 7B 63 67 05 EF \r7BB 22 90 FF FF 8F FF 10 FF \r7BB 23 FF FF FF FF FF FF FF \r7BB 24 FF 46 F0 71 6E 00 FF \r7BB 25 FF 00 FF FF 00 00 00 \r\r>'
2019-03-10 14:48:35,844 [salt.loaded.ext.engines.obd_manager:104 ][DEBUG   ][822] Got query result: 9.5
```

Maybe there is a clue there `Changed header from '7DF' to '7B3'`

alklnofx, 2019-03-23:
12 means subfunction is not supported.
So when the PID is correct you probaply have to change to a other session
first to read out this data...

Is in GDS a command to turn on the heating/aircon?

gtrdriver, 2019-03-23:
Hi

i dont get any data with the Header "7BB". I just wrote a small script wich test a PID Range and check result perhaps there is a other pid here with this header ...

I will access the GDS next week i will look what i can find ...

Niponyoshi, 2019-03-23:
@gtrdriver
How do you send command to mysql ?
I have personnel project I can share. 
It is run on my personnal cloud. 
Can you help me to write the right command from my autopi to mysql data base ?

gtrdriver, 2019-03-23:
Hi

as i wrote here im not using autopi - it uses a common Raspberry with a Common OBD/USB Adapter.

With python its realy simple - you open comport - itit the adapter - then send pid command: (example: 220101" - read the Imput from the Serial device, decode the hex and collect it
After each read sesson of all pids i send the results to a mysql Database.

its all written in python and - be aware - im not a python god ....

it works well - but not more ...

gtrdriver, 2019-03-23:
@all   

No Luck - searchd a bit range of pids for this header but no luck ...

Also with tourqe Pro i dont get valid values on 220100 ....

Perhaps its a country specific Problem or it depend on the used OBD Adapter ...

plord12, 2019-03-23:
Oh dear, sorry about that.  

In the past, with torque pro, a generic bluetooth OBD adapter, the csv files here and my kona switch fully on I did see the temperature -

[image]

Strange that you don't see the same.

plord12, 2019-03-26:
> Waiting for my autopi, and in the meantime, trying to get Navdy's OBD to work with Kona EV.
> Does anyone know whether Kona outputs the speed PID, and if yes, what the speed PID is?
> 
> It seems like it isn't returning anything on the usual PID 13.

Yea, it looks to me that its PID 220100, 7B3, ad.  So in torque pro format -

`Vehicle_Speed,Vehicle Speed,220100,ad,0,147,km/h,7B3`

The speed and rpm are correlected as you would expected -

[image]

I've added to autopi library.

orcinus, 2019-03-27:
Awesome! Thanks!

plord12, 2019-04-03:
> I will access the GDS next week i will look what i can find ...

Did you get anywhere with this ?

Thanks,

Pete

plord12, 2019-04-27:
I just noticed that the OVMS folks have been making progress with the e-Nero ( see https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/tree/master/vehicle/OVMS.V3/components/vehicle_kianiroev/src ).  Notably tyre pressures, temperators and display odometer.

I can no-longer use torque pro, but guessing the config, it should be something like -

```
001_Tire Pressure Front Left,Tire Pre_FL,22C00B,e/5,0.00,120.00,psi,7A0
001_Tire Pressure Front Right,Tire Pre_FR,22C00B,i/5,0.00,120.00,psi,7A0
001_Tire Pressure Back Left,Tire Pre_BL,22C00B,q/5,0.00,120.00,psi,7A0
001_Tire Pressure Back Right,Tire Pre_BR,22C00B,m/5,0.00,120.00,psi,7A0

001_Tire Temperature Front Left,Tire Temp_FL,22C00B,f-50,-40,65,C,7A0
001_Tire Temperature Front Right,Tire Temp_FR,22C00B,j-50,-40,65,C,7A0
001_Tire Temperature Back Left,Tire Temp_BL,22C00B,r-50,-40,65,C,7A0
001_Tire Temperature Back Right,Tire Temp_BR,22C00B,n-50,-40,65,C,7A0

Odometer,Odometer,22B002,i*16777216+j*65536+k*256+l,0,200000,m,7C6
```

muhkuhgh, 2019-10-08:
This doesn't 100% perfectly fits here, but as I'm totally new to this and still a little confused aber reading... may I just quickly ask: Does the AutoPi Stick (2nd Generation) works with the Hyundai Kona EV when I feed him these PIDs and I can log Voltage, Current and so on?

plord12, 2019-10-08:
Short answer yes.  The PID's here have been added to the auopi community library ( as Kona EV 2018 ).

However, you might also want to checkout https://www.openvehicles.com - I see that supports the Kona now.