JejuSoul/OBD-PIDs-for-HKMC-EVs #27: Can't import Niro PIDs to Torque

kardaman, 2018-12-16:
Hello

After downloading the PIDs CVS Files via right click on the raw button and transferring them over USB cable to my phone I can import the 002_Kona&Niro_EV_HVAC.csv File and find the two PIDs in the list afterwards, like it should be.
Doing the same with 003_Kona&Niro_EV_BMS.csv and ABRP_Kona&Niro_PIDs.csv , no new PIDs appear.
What do I wrong?
I would like to share my data to improve ABRP for the Niro.
Thanks

plord12, 2018-12-16:
I understand ABRP_Kona&Niro_PIDs.csv is for a better route planner.

I just tried clearing and re-add 003_Kona&Niro_EV_BMS.csv  ( Add predefined set ) and I get a long list of PID's.

[image]

kardaman, 2018-12-16:
Hello

I solved the problem. Had a wrong/old version of Torque installed. Deinstall and reinstall Torque solved all problems.

Thanks

closed 2018-12-16 by kardaman