JejuSoul/OBD-PIDs-for-HKMC-EVs #29: Ioniq formula for actual energy charged?

ojherbst, 2019-01-06:
Is there is any way to calculate the actual energy charged at a single charging session in Torque? This should possible based on the difference between 'Cumulative Energy Charged' before charging and 'Cumulative Energy Charged' after charging.

migle, 2019-01-06:
Yes, that's the right way. When using Synchronous logging that difference is within 1% of the area below the Battery power curve, and with most fast chargers, except those that have a very long cable, within 1% of what is computed by the charger.

It is not equivalent to the energy computed by an EVSE, because there is power dissipation at the OBC.

ojherbst, 2019-01-11:
Thanks, that makes sense. I checked (CEC 2451 kWh, CED 2360 kWh) and after clearing the current state of charge (16.2 kWh calc from SOC Display) my car has already an accumulated energy dissipation of approximately of 74.8 kWh or 3% over its relatively short lifetime.

I could just integrate the difference of 3% for my car but this would be a messy and very inaccurate calculation. Thus, I was experimenting with a formula (see below) using a rolling average RAVG val{value} but while the formula works, I can't figure out which reset command to use and wasn't brave enough just to press all of them. The idea of using RAVG val{value} was to reset a calculated value of it to 0.0 kWh before charging and then let it just resume calculating from there. But again, I don't know how to reset it.

Any ideas on how to reset it or how I could capture the value displayed for CEC before charging that would be great. 

004_CALC CEC now,CEC now,2101,RAVG(val{000_Cumulative Energy Charged}),0,1000000,kWh,7E4

migle, 2019-01-11:
> after clearing the current state of charge (16.2 kWh calc from SOC Display) my car has already an accumulated energy dissipation of approximately of 74.8 kWh or 3% over its relatively short lifetime.

So, you think in the beginning, when CEC=CED=0, SoC was at 0%? I can't tell anymore. I have CEC=8028.2 CED=7807.4. 3% is reasonable, I get to the same number. That means >97% energy efficiency for the battery, it's plausible.

> I could just integrate the difference of 3% for my car but this would be a messy and very inaccurate calculation. 

Yes. What are you trying to do? You want to present State of Charge, but an absolute one, not %, but kWh?

You can't really get around that magic 3% number, unless you do something much more complicated. Li-Ion batteries have very high coulombic efficiency, >99%, that means that you get back almost all the charge you put there. But charge is electrical charge related to the number of ions, measured in Coulombs or Ampère-hours, not energy. If you check the other two counters, CCC and CDC, you'll get a much closer number. Mines are CCC=22285.2Ah, CDC=22279.2Ah, indicating >99,9% coulombic efficiency.

So, CCC and CDC will give you a much more trustworthy measure of absolute State of Charge in Ah. The amount of energy those Ah will yield will vary with the current. A very high charge current will raise the voltage a lot more than a moderate one, thus requiring more energy for the same amount of charge. A very high discharge current will lower the voltage a lot more too, thus yielding less energy.

All in all, this means that if your foot is heavy (current is high), or if the temperature is low or if the SoC is low (internal resistance rises), the same amount of charge will yield less energy than it would usually. For instance, I rely on 27,6kWh when driving moderately in summer, but only on 27,0kWh if I'll be driving on the motorway.

So, if presenting absolute SoC is your goal, it will be too optimistic at the times where you need it the most. Those 3% are probably a little less when everything is fine, and a little more than 3% with speed, cold or low SoC.

If you use CCC and CDC, on the other hand, because it's energy that you need to know, you'll need to know which will be the expected average discharge voltage from the current SoC down... you can have that curve tabled.... But you probably see that you are not in position of giving a better estimate than the BMS gives.

Over time, I ploted energy discharged - energy charged versus SoC in %, and it is such a good straight line... We can't beat that.

If you trust the BMS, on the other hand, you can just compute the absolute SoC like this kWh = %SoC * %SoH * 28 (or maybe a smaller number, to be conservative).

I don't know if this was what you were trying to do, my english... And maybe you know all this a lot better than I do.
On the other hand, these other questions:

> I don't know how to reset it.

I don't either, I don't care that much about Torque. Maybe ask that in the Torque Pro forums. Maybe someone else here knows.

> how I could capture the value displayed for CEC before charging that would be great

Personally, I log data to a file. When driving, with kids in the car, I don't really have time to use the phone. I just log data all the time and have the logs uploaded to my nextcloud automatically. Eventually, if I have the time, it's easy to see a charging session in the logs and what values CEC and CED were at before and after. Log CEC, CED, HV Charging, Rapid Port, Normal Port, for that... maybe also CCC, CDC, DC Voltage, Current, max and min temperature, max and min cell voltage, max and min cell deterioration.

The car stops when min voltage is 3,0V regardless of whether charge could be extracted from the battery or not. So, if you push it at a low SoC, the car can stop before you reach 0%.