JejuSoul/OBD-PIDs-for-HKMC-EVs #35: Kona EV - HVAC - "NO DATA" and Charging dont diaply

gtrdriver, 2019-03-16:
Hello

First of all - Great work !!! -thank you ...
Im using Torque Pro for testing the PID Sets form this github Site and use a RPI + a OBD Adapter to read /decode and send this data to a Database Server for Monitoring ...

I have tested all Values i need for my projekt but his 3 Values dont work here:

1: HVAC:
Sending the PID "220100" deliver "no Data" - (on Torque or on a Console)

2: I want a Binary to dermine the car is Charging (AC or DC)
The Pro Configured Set "000_HV_Charging" dont work here ..

is there any known issue regarding this issue ?

migle, 2019-03-17:
Regarding your second point, besides 000_HV Charging, you have 000_Normal Charge Port and 000_Rapid Charge Port, which should make it clear whether it is AC or DC.

premultiply, 2019-03-18:
It is 0x220101 on 0x7e4

gtrdriver, 2019-03-18:
Hi

first of all tank you for reply

000_Normal Charge Port allways report "0" at Tourqe
000_Rapid Charge Port  i have tested yesterday and it look like this pid works...
000_HV Charging allways report "9"

premultiply, 2019-03-18:
How can a single bit show a value of „9“?

gtrdriver, 2019-03-18:
Hi

sorry - i mean "0" ....

plord12, 2019-03-20:
Yea, there is something not quite right here.  On my Kona I see -

* "j" - always 0x03
* "ay" - 0x0D when driving, 0x09 when charging and sometimes 0x01 ( I think this is car on but neither charging nor driving )

piec2, 2019-04-16:
I still haven't figured out the Normal Charge Port. Nothing works for it reliably so right now I'm considering using the following formula :
val{000_BMS Main Relay}-val{000_BMS Ignition}

That seems to work for me. 
When we drive, both are '1' of course. 
When we recharge, the vehicle ignition is Off (don't know if you can try to recharge with the Ignition On).
The BMS Main relay is always ' 1' when recharging.

Have you found anything different?

JejuSoul, 2019-07-13:
Am closing this - will open 2 new threads 
1 about charging
2 about HVAC

closed 2019-07-13 by JejuSoul