JejuSoul/OBD-PIDs-for-HKMC-EVs #38: Correction in BSO Display & SOH for Ioniq PHEV

Wekoyote, 2019-05-06:
Hi,

I have found that the file Hyundai_Ioniq_EV_BMS_data.csv, which is the one used for Ioniq Plug-in, does not work SOC Display and SOH, since it gives erroneous values.

The changes to be made are in the last lines of the file and are the following:

**SOC Display:**
_000_State of Charge Display,SOC Display,2105,**af**/2,0,100,%, 7E4_ 
to this line
_000_State of Charge Display,SOC Display,2105,**am**/2,0,100,%,7E4_

**SOH:**
_000_State of Health, SOH, 2105,((**z**<8)+**aa**)/10.0,100,%,7E4_
to this line
_000_State of Health,SOH,2105,((**aj**<8)+**ak**)/10.0,100,%,7E4_

[Hyundai_Ioniq_PHEV_BMS_data.csv.zip](https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/files/3150077/Hyundai_Ioniq_PHEV_BMS_data.csv.zip)

Attached modified file, and that I have allowed the name change to _Hyundai_Ioniq_PHEV_BMS_data.csv_

Regards,

JejuSoul, 2019-07-11:
Thanks Wekoyote. I have added a folder for the Ioniq PHEV.
The battery should be the same as the Kia Niro PHEV.

JejuSoul, 2019-07-11:
see - [Comparing Hyundai / Kia Electric Vehicle Batteries](url=http://www.mykiasoulev.com/forum/viewtopic.php?f=6&t=944&p=8200#p8200)

closed 2019-07-12 by JejuSoul