JejuSoul/OBD-PIDs-for-HKMC-EVs #39: Charging Flags on the 64kWh Kona / Niro Soul EVs

JejuSoul, 2019-07-13:
-
Much discussion about these flags in previous issues.
000_Normal Charge Port 
000_Rapid Charge Port 
000_HV Charging

The best analysis is by piec2  here - https://docs.google.com/spreadsheets/d/1JyJnXh7DOvzTl0cbWZRpW9qB_OgEOM-w_XOiAY_a64o/edit?usp=sharing

It seems the best way to calculate charging is to calculate val{000_BMS Main Relay}-val{000_BMS Ignition}

Can someone post here 3 working codes.

piec2, 2019-09-09:
I've done further testing, both on Level 2 and Level 3 charging on the Kona EV (including using the Utility Mode of the Kona - aka Camping Mode) and I confirm that the 000_HV Charging is correctly represented by: val{000_BMS Main Relay}-val{000_BMS Ignition}

I've tested all 8 bits of 0x220101 {ay:x}  as well as all 8 bits of 0x220101 {j:x} and neither of them can indicate if we are charging over the normal charge port, or the rapid charge port.

One down, 2 to go!

JejuSoul, 2019-09-17:
Thanks for testing this. I have added 000_HV Charging to the 64kWh Soul EV file.
I am assuming the PIDs for the 64kWh Soul EV are identical to the  64kWh  Niro EV.

There is a separate  64kWh  Kona / Niro EV file which I haven't changed.

Krolow93, 2020-12-20:
@JejuSoul I might have found some useful info for one of these flags.

`000_HV_Charging,Charging,0x220101,{F:4} ,0,1,,7E4`
Seems to do the trick or am I missing something? 
It returns 0 all the time except when Charging with AC or DC from my testing.
If it makes any difference this is on a 2020 Kona 64kw and the charge plugs are type 2 and CCS for AC and DC respectively.

I did not make the discovery but found it on: https://forum.abetterrouteplanner.com/topic/1667-kona-2020-wrong-charging-status/

I don't understand these codes nearly as good as you lot so might be useful info in that post or might already be in the analysis @piec2 has done before?