JejuSoul/OBD-PIDs-for-HKMC-EVs #41: Open Source Analysis of Hyundai / Kia CAN messages, and other Programs using them.

JejuSoul, 2019-07-13:
Here are some sites with Open Source Analysis of Hyundai / Kia CAN messages.

27kWh Kia Soul EV - https://docs.google.com/spreadsheets/d/1YYlZ-IcTQlz-LzaYkHO-7a4SFM8QYs2BGNXiSU5_EwI/edit?pref=2&pli=1#gid=0

64kWh Kia Niro EV - https://docs.google.com/spreadsheets/d/1eT2R8hmsD1hC__9LtnkZ3eDjLcdib9JR-3Myc97jy8M/edit#gid=587742707

28kWh Hyundai Ioniq Electric - https://docs.google.com/spreadsheets/d/1R2dW_4-ANg04TQdkHH52-8wdYItBkWDfppcsqOc-m1M/edit?usp=sharing

Does anyone know any more? Are there extra values seen on these sites that we can use here?

birkir, 2019-07-19:
Nice find! For the Kia Niro, Hyundai Kona would also apply, right?

Where would you have to plug into the CAN bus to get these kinds of information?

sunnyhaibin, 2023-01-17:
@JejuSoul Hello! In the "command" spreadsheets, how did you send those commands and have the car accepted the commands in 0x770? I'm trying this out in the 2020 Hyundai Sonata, which the messages are almost identical to the Kia Niro EV except the EV related messages.

I entered Diagnostic Session via \x10\x03, received positive response \x50\x03, then tried to send the commands to 0x770 as mentioned [here](https://docs.google.com/spreadsheets/d/1eT2R8hmsD1hC__9LtnkZ3eDjLcdib9JR-3Myc97jy8M/edit#gid=1419959931), but I kept receiving negative response \x7F\x2F\x33 for all commands sent to this ECU after.

henkhoo, 2024-06-15:
Hi ,  i have a ( Niro HEV ) Hybrid 1,5Kw battery, spitting out the following HEx messages. looking for SOC , Volt and others.

Where to look for it ?  

22285  670  8  9E 7 34 8 A8 6D 4 0 
22286  51E  8  0 0 0 0 C 0 0 0 
22287  588  8  45 9 0 0 0 6D 0 0 
22288  5AD  8  B9 B9 0 0 14 14 0 55 
22289  5AE  8  ED 0 6D 13 C 0 0 0 
22290  5AF  8  0 0 0 0 0 0 0 0 
22301  5F1  8  A8 16 45 9 0 0 0 0

fish-face, 2024-10-07:
> @JejuSoul Hello! In the "command" spreadsheets, how did you send those commands and have the car accepted the commands in 0x770? I'm trying this out in the 2020 Hyundai Sonata, which the messages [...]

Not sure if we're looking at the same stuff or not, but I have the following sniff from a session with an ELM327 scanner:

```
-->  41 54 53 48 37 34 34 0d                           ATSH744.
<--  4f 4b 0d 0d 3e                                    OK..>
-->  31 30 30 33 31 0d                                 10031.
<--  37 34 43 30 36 35 30 30 33 30 30 33 32 30 31 46   74C065003003201F
-->  32 32 45 30 30 32 36 0d                           22E0026.
<--  37 34 43 31 30 32 33 36 32 45 30 30 32 46 46 46   74C102362E002FFF
<--  37 34 43 32 31 30 30 46 30 30 30 30 30 30 30 30   74C2100F00000000
<--  37 34 43 32 32 41 30 30 30 30 30 30 30 30 30 30   74C22A0000000000
<--  37 34 43 32 33 30 30 30 30 30 30 30 30 30 30 30   74C2300000000000
<--  37 34 43 32 34 30 30 30 30 30 30 30 30 39 39 30   74C2400000000990
<--  37 34 43 32 35 31 37 41 41 41 41 41 41 41 41 41   74C2517AAAAAAAAA
```
I saw you also talking about this magic string "1003" and wondered if you knew any more about it - I don't really understand that this is doing but I think 1003 is always sent before querying ECU 744.

Tuoris, 2025-02-11:
> [@JejuSoul](https://github.com/JejuSoul) Hello! In the "command" spreadsheets, how did you send those commands and have the car accepted the commands in 0x770? I'm trying this out in the 2020 [...]

To send the commands to the car I wrote a small app - https://tuoris.github.io/elm327_debugger/en , since Torque is not available on my smartphone.

I've just needed to set the protocol to ISO 15765-4 (`AT SP6`) and then sent `220101` to get some battery info. Should work for Huyndai Kona and Kia Niro. Most of th commands for the regular ICE cars (https://en.wikipedia.org/wiki/OBD-II_PIDs) will not work with EVs, to get the VIN for instance I've sent `1A80` instead of `0902`.