JejuSoul/OBD-PIDs-for-HKMC-EVs #43: Soul 2019 30kWh battery SOH shows "60% and 18kWh" whenever it drops below 110%

Bibadom, 2020-01-20:
The Kia Soul EV 2018-2019 30kWh actually have a 32kWh battery and a 110% SOH when brand new. When this SOH drops. Torque Pro start showing a SOH of 60% and 18kWh total capacity. It seems to be like this for every 30kWh models.

JejuSoul, 2020-01-21:
@Bibadom - no one has ever shown data before for a 2018 or 2019 30kWh battery that differs from zero deterioration. Hence you are the first to report, which is why I am asking you for more details. 
Can you explain / show some data as to why you think the 30kWh actually has a 32kWh battery.
Why do you think your car has 110% SOH when brand new?
Can you show the actual data that Torque is reading from your car.

I don't have a 100 cell '30kWh' battery, but I recently had a battery replacement in my 2015 Soul EV.
I now have a new battery with the 2018 cells. It appears to be a 96 cell '30kWh' battery.
I am curious to know more about these cells.

pemessier, 2020-01-21:
See this Facebook thread with more details about this issue: https://www.facebook.com/groups/KiaSoulEVQuebec/permalink/3176822895680754/

The main issue is that min det reports 100%, while max det reports 0%. Something is probably wrong in the min/max values decoding.

I'll check with @Bibadom to get all the raw data on his car.

Bibadom, 2020-01-21:
I don't think I have saved the past reading but the fact I was thinking
it's a 32kWh battery come from it's the reading I had on Torque Pro since I
have the car. Before the last one witch show now 18kWh. And from what I
understand from the talking on the facebook group, I'm not the only one who
had these readings

Luc Paradis
514-965-9804
lucparadis888@gmail.com

Bibadom, 2020-01-21:
tu peux m'envoyer la procédure par ici aussi

Luc Paradis
514-965-9804
lucparadis888@gmail.com

pemessier, 2020-01-23:
Sorry for the late answer.

The easiest way to get the raw data is using an OBD2 terminal app. The one I used in the past no longer exist on Google Play. I tried several today, and this one was easy to use:

https://play.google.com/store/apps/details?id=com.glmsoftware.OBDNowTerminal&hl=en_CA

Here's the procedure:

1. Make sure to connect/power on your OBD2 adapter just before connecting using OBD Now Terminal (do not connect using Torque before, for example).
1. Open OBD Now Terminal, and accept the filesystem access permission if needed
1. Connect to your OBD2 adapter using the app
1. Send the following commands in the following order
    * ATH1
    * 2101
    * 2102
    * 2103
    * 2104
    * 2105
1. This should look like this:
[image]
1. The data was logged in a file. The path on the Android device is given in OBD Now Terminal help, but for reference it's under /storage/emulated/0/OBD Now Terminal/Log***.txt
1. Upload the log file here please. Thanks.

If you need help do not hesitate to contact me (you can reach me on FB Messenger if you like).

Bibadom, 2020-01-23:
>    2. Open OBD Now Terminal, and accept the filesystem access permission
>    3. Connect to your OBD2 adapter using the app
>    4. Send the following commands in the following order
>    5. This should look like this:
>    6. The data was logged in a file. The path on the Android device is
>    7. Upload the log file here please. Thanks.

Voilà

Luc Paradis
514-965-9804
lucparadis888@gmail.com

  1 Connection: 23 janvier 2020 18:39:03    	
  2 OBD Now Terminal - Version 1.0.16       	39:04.005
  3 Android Version: 9                      	39:04. 014
  4 Build Number: PKQ1.190522.001           	39:04.014
  5 Manufacturer: LGE                       	39:04.014
  6 Model: LG-H873                          	39:04.014
  7 Brand: lge                              	39:04.014
  8 Device: lucye                           	39:04.015
  9 Product: lucye_global_ca                	39:04.015
 10 Country: Canada                         	39:04.019
 11 Bluetooth Device: OBDII                 	39:04.582
 12 Bluetooth Address: 00:1D:A5:00:1E:38    	39:04.584
 13 ATH1                                    	39:29.347
 14 ATH1                                    	39:29.472
 15 OK                                      	39:29.472
 16 >                                       	39:29.472
 17 2101                                    	39:49.782
 18 2101                                    	39:49.966
 19 7EC 10 3D 61 01 FF FF FF FF             	39:49.967
 20 7EA 10 0E 61 01 F0 00 00 00             	39:49.977
 21 7EC 21 93 18 84 23 28 03 00             	39:49.977
 22 7EA 21 E3 05 F8 02 00 00 00             	39:49.977
 23 7EC 22 01 0F 1F 05 03 05 04             	39:49.977
 24 7EA 22 00 00 00 00 00 00 00             	39:49.977
 25 7EC 23 04 04 04 00 05 C1 18             	39:49.977
 26 7EC 24 C1 14 00 00 8B 00 01             	39:49.977
 27 7EC 25 0E C3 00 01 0E F0 00             	39:49.977
 28 7EC 26 00 66 BD 00 00 63 CC             	39:49.977
 29 7EC 27 00 28 5A 17 2D 01 83             	39:49.977
 30 7EC 28 00 00 00 00 03 E8 00             	39:49.977
 31 >                                       	39:49.977
 32 2102                                    	39:53.581
 33 2102                                    	39:53.794
 34 7EC 10 26 61 02 FF FF FF FF             	39:53.795
 35 7EA 10 21 61 02 FF FF 80 00             	39:53.795
 36 7EC 21 C1 C1 C1 C1 C1 C1 C1             	39:53.795
 37 7EA 21 8E C1 00 00 00 00 00             	39:53.795
 38 7EC 22 C1 C1 C1 C1 C1 C1 C1             	39:53.795
 39 7EA 22 00 00 00 04 00 00 00             	39:53.796
 40 7EC 23 C1 C1 C1 C1 C1 C1 C1             	39:53.796
 41 7EA 23 00 00 1C 00 30 2E 2C             	39:53.796
 42 7EC 24 C1 C1 C1 C1 C1 C1 C1             	39:53.796
 43 7EA 24 08 08 07 08 66 52 00             	39:53.796
 44 7EC 25 C1 C1 C1 C1 00 00 00             	39:53.796
 45 >                                       	39:53.796
 46 2103                                    	39:58.628
 47 2103                                    	39:58.837
 48 7EC 10 26 61 03 FF FF FF FF             	39:58.838
 49 7EA 03 7F 21 12                         	39:58.838
 50 7EC 21 C1 C1 C1 C1 C1 C1 C1             	39:58.838
 51 7EC 22 C1 C1 C1 C1 C1 C1 C1             	39:58.838
 52 7EC 23 C1 C1 C1 C1 C1 C1 C1             	39:58.838
 53 7EC 24 C1 C1 C1 C1 C1 C1 C1             	39:58.838
 54 7EC 25 C1 C1 C1 C1 00 00 00             	39:58.838
 55 >                                       	39:58.839
 56 2104                                    	40:02.655
 57 2104                                    	40:02.844
 58 7EC 10 26 61 04 FF FF FF FF             	40:02.845
 59 7EA 03 7F 21 12                         	40:02.845
 60 7EC 21 C1 C1 C1 C1 C1 C1 C1             	40:02.845
 61 7EC 22 C1 C1 C1 C1 C1 C1 C1             	40:02.845
 62 7EC 23 C1 C1 C1 C1 C1 C1 C1             	40:02.845
 63 7EC 24 C1 C1 C1 C1 C1 C1 C1             	40:02.845
 64 7EC 25 C1 C1 C1 C1 00 00 00             	40:02.845
 65 >                                       	40:02.845
 66 2105                                    	40:05.911
 67 2105                                    	40:06.052
 68 7EC 10 2D 61 05 FF FF FF FF             	40:06.052
 69 7EA 03 7F 21 12                         	40:06.052
 70 7EC 21 C1 C1 C1 C1 C1 04 04             	40:06.053
 71 7EC 22 03 00 00 00 00 18 84             	40:06.053
 72 7EC 23 23 28 00 01 50 04 04             	40:06.053
 73 7EC 24 00 00 18 03 E8 19 9A             	40:06.053
 74 7EC 25 00 00 C1 C1 00 00 00             	40:06.053
 75 7EC 26 00 00 00 00 00 00 00             	40:06.053
 76 >                                       	40:06.053
 77 save                                    	46:34.249
 78 save                                    	46:34.500
 79 ?                                       	46:34.501
 80 >                                       	46:34.501

pemessier, 2020-02-10:
So the deterioration data is decoded correctly:
7EC 24 00 00 18 03 E8 19 9A

Max det = 0x0000 = 0%
Max det cell = 0x18 = 24

Min det = 0x03E8 = 100.0%
Min det cell = 0x19 = 25

This do not make any sense to me. This should be confirmed using Kia KDS in my opinion, to understand what values they get. This may be a bug in the BMS.

Both values should be quite low, as I understand your car is not too old.

This behavior was seen on another car as well (saw in the Facebook thread).

Anyone else has an explanation?

Spannungsfall, 2020-02-25:
Hello everybody,
I have the exact same problem and would like to help solve it. Here <https://www.facebook.com/groups/KiaSoulEVQuebec/?notif_id=1582549249271022&notif_t=group_r2j_approved>
I posted a video with my read data. Maybe someone will find the mistake ...

Thank you

JejuSoul, 2020-02-26:
This issue has been discussed before - https://www.mykiasoulev.com/forum/viewtopic.php?p=8171#p8171

The SOH readings given by Torque do not work correctly on the 30kWh versions of the Soul EV.

We are still waiting for a user with less than 100% SOH usable capacity to post their data.

Bibadom, 2020-02-26:
then how to explain in my first months having the car, the reading I had from TorquePro was 120% with 32kWh? I came up the issue after it stopped showing me that to show what we see now. At this point I was thinking it is because I no longer have an complete SOH.

pemessier, 2020-02-26:
The formula in the CSV files explains why we are getting those values. Using Soul EV 30kWh files:

004_CALC SOH Battery in kWh	SOH in kWh	2101	**30+val{004_CALC Lost of SOH}**
004_CALC Lost of SOH	Lost SOH	2101	**30*val{004_CALC AVRDET For SOH Divided 100}**
004_CALC AVRDET For SOH Divided 100	AVRDET3	2101	**val{004_CALC AVRDET For SOH}/100**
004_CALC AVRDET For SOH	Lost SOH	2101	**10-val{004_CALC AVRDET}**
004_CALC AVRDET	Average DET	2101	**val{004_CALC MaxMin DET}/2**
004_CALC MaxMin DET	AVRDET1	2101	**val{000_Maximum Deterioration}+val{000_Minimum Deterioration}**

To make it simple, all begins with the average of minimum and maximum deterioration values, given in %. If you put every equations together you get

SOH estimated in kWh = 30+(30*((10-(((Maximum Deterioration)+(Minimum Deterioration))/2))/100))

If you put MinDet=0 and MaxDet=0 in this formula, you'll get 33kWh.
In your case, **MinDet=100** and MaxDet = 0, and this is why you are getting 18kWh.

The calculation is wrong because the **minimum** deterioration is at 100% which makes no sense, but this is the value we are getting from the BMS so we are stuck with it.

JejuSoul, 2020-02-27:
Bibadom - if you follow the link I posted in the last comment you will see that this confusion was already known about in May 2018.

> We have conflicting data on the three 2018s we have tested.
> Ussiker and GizmoEV both see zero, zero for Min and Max deterioration . Siluzium sees zero, 100

What is new in your case is that the BMS has swapped from using one set of values to the other.
Do you have any idea what may have caused this to happen?

In both of these cases  zero, zero for Min and Max deterioration, or zero, 100 we assume that the actual usable capacity of the car is still 100%. No one has posted data to show how these values change once actual reduction of usable capacity occurs.

chrisli2018, 2020-03-14:
Same for me. Have a 2018 mod.
If you press "reset trip counters" it will show correct kWh for about 2 seconds, then it goes back to a lower number.

Spannungsfall, 2020-03-17:
> Same for me. Have a 2018 mod.
> If you press "reset trip counters" it will show correct kWh for about 2 seconds, then it goes back to a lower number.

You mean the "arrow-down" button?
I tried it yesterday, but the values doesn´t change...
Only after the first connection with OBD adapter, the values change to SoH 100% and 30kWh. But then quickly turn to 60% and 18kWh.

chrisli2018, 2020-03-17:
> You mean the "arrow-down" button?
> I tried it yesterday, but the values doesn´t change...
> Only after the first connection with OBD adapter, the values change to SoH 100% and 30kWh. But then quickly [...]

[image]

JejuSoul, 2020-03-17:
-
The latest data I have for a 30kWh Soul EV is from a June 2017 car in Norway with 47,000km.
The deterioration counters still both show zero.

Using Torque data we see that at SOC (display) = 100% on a 2018 after 47,000km
The cell voltages are 4.10V. Total battery voltage is 411.4V.
Battery temp at 12C.