JejuSoul/OBD-PIDs-for-HKMC-EVs #46: Ioniq HEV Extended PIDs don't work with the model 2020

nono031, 2020-08-12:
Hi,

I installed CSV from the repository and added into torque but I don't receive any value from those new PIDs

I also tried to use torquescan to find other PIDs but without result (I did a simple scan not the "full scan")

I want to help to improve the support of the ionic, do you have any guidelines ?

Best,

Arnaud

banf1she, 2021-05-11:
> 
> 
> The Ioniq facelift shares the same EV electronics as the Kona. The original Ioniq PIDs wont work, but those for the Kona/Niro will.

I will look into this and report back, thank you.

banf1she, 2021-05-19:
I have found that it indeed does share many of the PIDs from the Kona EV, however, the VMCU PIDs for the Kona in this repo do not appear to work. I can include the logs from Torque Scan, would that help in narrowing them down?