JejuSoul/OBD-PIDs-for-HKMC-EVs #65: PID Mapping for new Gen of Tucson/Sportage HEV / PHEV

Ryccoo, 2024-05-01:
Hi, I am just checking if someone tried to reuse some of the PIDs for new generation of the hybrid / plug in hybrid Hyundai Tucson or Kia Sportage.

I have a 2023 HEV Tucson and was curious what could I get out of it. I know there is a lot of possibilities (used Car scanner app) now I am just looking for PIDs to use arduino to read few of them.

girishadat, 2025-01-25:
Looking for the same, but on a 2019-20 Tucson 48v mildhybrid.

Did you manage to find any? @Ryccoo

theroof24, 2026-09-02:
Hello, I have a 2024 PHEV and the CARSCANNER app has some of them in it. I can see charge cycles, real time current and voltage. Trying to find a way to get PIDS out of the app that I can use for some custom code.